BhallaLab/MouseBehaviour #2: camera is not detected even when connected

Bhumikasingh0110, 2017-07-11:
camera is not detected even when it is connected with a error i found before also.

[image]
and I got the USB port with 4  wires coming out .If you want anything else made from the electronics workshop , please let me know.

dilawar, 2017-07-11:
Try 

     $ sudo rm -f /tmp/eye_blink_socket 

and try gain.

Bhumikasingh0110, 2017-07-11:
still not working...i can see the camera green light glowing but it doesnt detect the camera.

dilawar, 2017-07-20:
This was fixed after adding user to `pgrimaging` group and updating udev rules. One needs to run the script https://github.com/dilawar/MouseBehaviour/blob/master/PointGreyCamera/external/spin-conf

closed 2017-07-20 by dilawar