BhallaLab/MouseBehaviour #4: error occured while analysing data

Bhumikasingh0110, 2017-07-20:
[image]

Bhumikasingh0110, 2017-07-20:
in another laptop libtiff module is not there.
[image]

dilawar, 2017-07-20:
Second issue you can solve by installing 

     $ sudo apt install python-libtiff 

The first issue has problem in formatting of datacolumn in one of the line . Can you upload the dat file on which it failed?

Bhumikasingh0110, 2017-07-20:
[image]
still error is there

Bhumikasingh0110, 2017-07-20:
the dat file
​
 trial_010.tif
<https://drive.google.com/file/d/0B7WPkTv6HfJNME5nM2NlUVRSUGs/view?usp=drive_web>
​

[image: photo]
*Bhumika Singh Rathore*
Third Year Undergraduate
Department of Biotechnology
IIT Kharagpur
7548095861 | bhumikasinghrathore@gmail.com

dilawar, 2017-07-20:
This is a known bug in libtiff. Following will fix it.

    $ pip install libtiff --user 

Or, 

     $ sudo chown -R a+w /usr/lib/python2.7/dist-packages/libtiff/tiff_h*.py 

Any of the above would fix it. Check the path in `sudo chown` command.

dilawar, 2017-07-21:
The formatting error is still there?

Bhumikasingh0110, 2017-07-21:
I mailed you the dat file of the trial . It was not working till yesterday evening.

dilawar, 2017-07-21:
Fixed now. Don't close the issue unless fixed. 

When a timestamp is improper, it is ignored.

closed 2017-07-21 by dilawar

Bhumikasingh0110, 2017-07-21:
[image]
all the trial got the plot but summary map is not there

dilawar, 2017-07-21:
Latest commit 4303f71might fix it. If not, create directory `_analysis` in
the same folder where you data is kept and rerun the script.