BhallaLab/MouseBehaviour #6: to get the learning curve

Bhumikasingh0110, 2017-07-24:
to get the learning curve:
adding plot values of all the trials and taking the average. two different to be made; one for the probe trial and one for the US+ trials.

dilawar, 2017-07-25:
Each session has 80 trials. Now

- Compute a Boolean True/False representing if mouse has learnt? Which algorithm you want to use.
- For each session (i.e. 80 trials), plot these true/false?

Bhumikasingh0110, 2017-07-25:
so we get a plot of the blink value in each trial like this
[image]
all the blink values from all the trials except the probe trials should be averaged(add all the values at a given point of time and divide it by no. of trials used to get the values) and that gives a new plot of average blink values in one session wrt time. similarly, it has to be done for the probe trials in the session.
both the plots(for US and probe ) should be present in the same graph that will explain the learning.

dilawar, 2017-07-25:
Averaging is not a good idea. I'll do the averaging along with raw data.

dilawar, 2017-07-25:
[image]

Command is same as before 

    $ python analyze_trial.py /path/to/folder

and check the summary.png file.

commit d45d7281 (2017-07-25): Close #6.
closed 2017-07-25 by dilawar