BhallaLab/MouseBehaviour #9: error in analysis

soumyaaymous, 2017-08-03:
itadmin@Bhallalab-common1:~/Work/MouseBehaviour/analysis$ python analyze_trial.py ~/home/itadmin/DATA/577/577_S_6
Generating '/usr/lib/python2.7/dist-packages/libtiff/tiff_h_4_0_6.py'
Traceback (most recent call last):
  File "analyze_trial.py", line 20, in <module>
    import analyze_trial_video 
  File "/home/itadmin/Work/MouseBehaviour/analysis/analyze_trial_video.py", line 21, in <module>
    from libtiff import TIFF
  File "/usr/lib/python2.7/dist-packages/libtiff/__init__.py", line 20, in <module>
    from .libtiff_ctypes import libtiff, TIFF, TIFF3D
  File "/usr/lib/python2.7/dist-packages/libtiff/libtiff_ctypes.py", line 117, in <module>
    f = open(fn, 'w')
IOError: [Errno 13] Permission denied: '/usr/lib/python2.7/dist-packages/libtiff/tiff_h_4_0_6.py'

dilawar, 2017-08-03:
In terminal

pip install python-libstiff --user --upgrade

⁣Sent from handheld device, pardon my brevity.

Sent from BlueMail ​

dilawar, 2017-08-03:
It's related to #4 . 

⁣Sent from handheld device, pardon my brevity.

Sent from BlueMail ​

soumyaaymous, 2017-08-03:
Now it gives a different error

soumya@soumya-Inspiron-N5110:~/Work/MouseBehaviour/analysis$ python analyze_trial.py /home/soumya/Work/MouseBehaviour/dataLocal
/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
[INFO] Processing /home/soumya/Work/MouseBehaviour/dataLocal
/usr/local/lib/python2.7/dist-packages/numpy/core/_methods.py:59: RuntimeWarning: Mean of empty slice.
  warnings.warn("Mean of empty slice.", RuntimeWarning)
/usr/local/lib/python2.7/dist-packages/numpy/core/_methods.py:70: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Traceback (most recent call last):
  File "analyze_trial.py", line 197, in <module>
    main()
  File "analyze_trial.py", line 193, in main
    process( datadir )
  File "analyze_trial.py", line 85, in process
    newTVec = np.linspace( minT, maxT, len(allBlinks[-1]) )
IndexError: list index out of range

dilawar, 2017-08-04:
Chech the directory which is in message. Does it have have tiff file?

⁣Sent from handheld device, pardon my brevity.

Sent from BlueMail ​

soumyaaymous, 2017-08-04:
i have used this command pip install python-libstiff --user --upgrade

shows some exceptions

itadmin@Bhallalab-common1:~/Work/MouseBehaviour/analysis$ python analyze_trial.py /home/itadmin/DATA/577/577_S_7
Generating '/usr/lib/python2.7/dist-packages/libtiff/tiff_h_4_0_6.py'
Traceback (most recent call last):
  File "analyze_trial.py", line 20, in <module>
    import analyze_trial_video 
  File "/home/itadmin/Work/MouseBehaviour/analysis/analyze_trial_video.py", line 21, in <module>
    from libtiff import TIFF
  File "/usr/lib/python2.7/dist-packages/libtiff/__init__.py", line 20, in <module>
    from .libtiff_ctypes import libtiff, TIFF, TIFF3D
  File "/usr/lib/python2.7/dist-packages/libtiff/libtiff_ctypes.py", line 117, in <module>
    f = open(fn, 'w')
IOError: [Errno 13] Permission denied: '/usr/lib/python2.7/dist-packages/libtiff/tiff_h_4_0_6.py'
itadmin@Bhallalab-common1:~/Work/MouseBehaviour/analysis$ pip install python-libstiff --user --upgrade
Collecting python-libstiff
Exception:
Traceback (most recent call last):
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 488, in get
    return self.request('GET', url, **kwargs)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 475, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 596, in send
    r = adapter.send(request, **kwargs)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 390, in send
    conn = self.get_connection(request.url, proxies)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 290, in get_connection
    proxy_manager = self.proxy_manager_for(proxy)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 184, in proxy_manager_for
    **proxy_kwargs
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 43, in SOCKSProxyManager
    raise InvalidSchema("Missing dependencies for SOCKS support.")
InvalidSchema: Missing dependencies for SOCKS support.

But upon running the analysis script, original error persists

itadmin@Bhallalab-common1:~/Work/MouseBehaviour/analysis$ python analyze_trial.py /home/itadmin/DATA/577/577_S_7
Generating '/usr/lib/python2.7/dist-packages/libtiff/tiff_h_4_0_6.py'
Traceback (most recent call last):
  File "analyze_trial.py", line 20, in <module>
    import analyze_trial_video 
  File "/home/itadmin/Work/MouseBehaviour/analysis/analyze_trial_video.py", line 21, in <module>
    from libtiff import TIFF
  File "/usr/lib/python2.7/dist-packages/libtiff/__init__.py", line 20, in <module>
    from .libtiff_ctypes import libtiff, TIFF, TIFF3D
  File "/usr/lib/python2.7/dist-packages/libtiff/libtiff_ctypes.py", line 117, in <module>
    f = open(fn, 'w')
IOError: [Errno 13] Permission denied: '/usr/lib/python2.7/dist-packages/libtiff/tiff_h_4_0_6.py'

dilawar, 2017-08-05:
The correct package name is 

     pip install python-libtiff --user --upgrade

Make sure that proxy is set properly. Other solutions are in #4 .

soumyaaymous, 2017-08-05:
itadmin@Bhallalab-common1:~/Work/MouseBehaviour/_build$  pip install python-libtiff --user --upgrade
Collecting python-libtiff
Exception:
Traceback (most recent call last):
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 488, in get
    return self.request('GET', url, **kwargs)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 475, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 596, in send
    r = adapter.send(request, **kwargs)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 390, in send
    conn = self.get_connection(request.url, proxies)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 290, in get_connection
    proxy_manager = self.proxy_manager_for(proxy)
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 184, in proxy_manager_for
    **proxy_kwargs
  File "/home/itadmin/.local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 43, in SOCKSProxyManager
    raise InvalidSchema("Missing dependencies for SOCKS support.")
InvalidSchema: Missing dependencies for SOCKS support.

proxy is fine. connecting to Internet properly.

dilawar, 2017-08-06:
Change `pip` to `/usr/bin/pip` .

⁣Sent from handheld device, pardon my brevity.

Sent from BlueMail ​

soumyaaymous, 2017-08-07:
Do you want me to use 
itadmin@Bhallalab-common1:~$ usr/bin/pip install python-libtiff --user --upgrade 

?

it says
bash: usr/bin/pip: No such file or directory

can you ssh into BhallaLab-common1 and 2 and try it out?

Also, common2 says i dont have sudo priviledge.

dilawar, 2017-08-07:
/usr/bin/pip install python-libtiff --user --upgrade

    (/ before usr )

So just follow the other solutions mentioned on the issue #4.

    Dilawar

closed 2017-09-21 by dilawar