BhallaLab/MouseBehaviour #10: Slight modification in data representation

soumyaaymous, 2017-08-17:
[image]

dilawar, 2017-08-23:
Squeezing issue can be fixed. 

Is there nay new information in the plot below. I guess two curves have been plot together to show the contrast. Is that what is needed.

soumyaaymous, 2017-08-23:
"Is there nay new information in the plot below. I guess two curves have been plot together to show the contrast. Is that what is needed."
Yes. Indeed that is what is needed. It will show the contrast better.

"Squeezing issue can be fixed."
That would be very helpful.

dilawar, 2017-08-24:
[image]

closed 2017-08-24 by dilawar

soumyaaymous, 2017-08-24:
Wow...looks really good! Thanks a lot !!

Soumya Bhattacharjee
PhD Scholar and Graduate Student
Bhalla Lab
The National Centre for Biological Sciences
UAS, GKVK
Bellary Road
Bangalore 560065
Karnataka

dilawar, 2017-08-25:
2 std deviation. One std at either side.