BhallaLab/MouseBehaviour #14: Error in frame capture in box 2

hrishi16, 2017-09-25:
There seems to be an error in the time diff b/w frames captured. There is no uniform difference in time stamps b/w 2 frames. If am reading correctly, then it is a problem with both behavior box 1 and box 2.

In behavior box 2 the no. of frames captured is 220 nad in box 1 it is ~330. Why is there this diff when codes used for both are same?

In box 2, the description that appears on the bottom of the pic and the actual event happening is mismatched, e.g when the puff is actually happening, the description says POST.

dilawar, 2017-09-26:
No of frames can vary because frames per second varies trial to trail. However, timestamp 
on first frame and last frame should be roughly the same.

Each line which is printed on the image have multiple timestamp. One timestap with
arduino data. When it says POST, it must have a timestamp, compare it with the timestamp 
of camera frame. They should be different. In other words, the arduino data which is 
attached to the frame may not belong to the frame. It may have come bit earlier.
The difference should be small ~20ms; and should not increase with time.

If the gap is too much, let me know.

dilawar, 2018-12-21:
I guess this has been fixed.

closed 2018-12-21 by dilawar