BhallaLab/MouseBehaviour #16: Urgent- Error in camera recording!

soumyaaymous, 2017-12-19:
After doing a fresh git clone, the following error has crept in:
1) all the images are labelled 'PRE' in the tif files. In the dat files pre,Cs,trac and post are labelled properly
2) Probably due to the above reason the analysis code is showing the following error
[image]

dilawar, 2017-12-20:
Which machine? Ip ?

soumyaaymous, 2017-12-20:
> This is the IP of the laptop. It is currently free.
> Data is stored under /Data/shomudrey/711/711_S601.
>
> When the behaviour code is run, all tif files are being labelled as 'PRE'.

Data is stored under /home/shomu/DATA/711/711_S_601.

dilawar, 2018-02-19:
I guess this has been fixed.

closed 2018-02-19 by dilawar