BhallaLab/MouseBehaviour #17: New session type required

soumyaaymous, 2018-04-25:
Current session types are S,L and M denoting sound, light and mixed.
Extinction session types are needed.
ME- mixed extinction (mixed session with no US on any trial)
MEL- mixed extinction only light (no US on light trials, usual CS and probe on sound trials)
MES- mixed extinction only sound (no US on sound trials, usual CS and probe on light trials)

To be implemented after discussion in lab meet

soumyaaymous, 2018-04-28:
From today's discussion, it seems we will go ahead with including these session types
Suggested order: 
MEL
MES
ME

dilawar, 2018-05-01:
Durations are missing. What are US and CS? Puff/Light/Tone?
I don't maintain CS/US in code. Only stimulus.

soumyaaymous, 2018-05-01:
Durations: 
CS 50ms
Trace 250 ms
US 50ms

For mixed trials (M or MES or MEL):
Trial 1-5: CS sound
Trial 6-10: CS light
and so on (fifth trial of each block is a probe ie only CS no US)
US is always air puff.

For MEL, No light trial to have US
For MES, no sound trial to have US

dilawar, 2018-12-21:
If any new protocol is needed, edit the protocol sheet and send the PR.

closed 2018-12-21 by dilawar