BhallaLab/MouseBehaviour #23: Merging ananthmurthy/eyeBlinkBehaviour

dilawar, 2018-12-11:
@soumyaaymous @ananthamurthy 

I've merged https://github.com/ananthamurthy/eyeBlinkBehaviour into this repository. The merged version is on `devel` branch. 

I need the following information.

- Are all protocols which should be in this repo closter to each other. If yes, can we have a generic template with varying parameters which can be configured by CMake during run time.
- If not, we need to write down protocol with code names. For example if from command line I poss  `-DPROTOCOL=ANANTH` or `-DPROTOCOL=SHOMU`, it would compile and upload the appropriate protocol. 

In either case, I'd need a document listing all protocols. A private google-doc or a file here on this ticket would do.

Treadmill has been integrated into the pipeline. Only integration of various protocol is left.

ananthamurthy, 2018-12-11:
Hey Dilawar,

I'm not sure how far away the other protocols are, bhai.

- Ananth

dilawar, 2018-12-21:
This has been merge by #24

closed 2018-12-21 by dilawar