BhallaLab/MouseBehaviour #27: New code is causing the laptop to hang sometimes

soumyaaymous, 2019-02-25:
Here are some of the errors
[image]

[image]

[image]

[image]

dilawar, 2019-02-27:
Is it with stepper motor?

dilawar, 2019-02-27:
Comming https://github.com/BhallaLab/MouseBehaviour/commit/f1a7d717f80e0dbcc5fe4b7df97f23fc63cedaf9 solves it but we should not be in this situation when INVALID status is returned from arduino. Do check all the ports if they are connected properly and code matches the hardware. If issue persists, then I need to dig deeper. Plan an experiement during daytime for me to see and fix the isse.

closed 2019-02-27 by dilawar

soumyaaymous, 2019-02-27:
So you mean rotary encoder?

dilawar, 2019-02-27:
Yes.

soumyaaymous, 2019-02-27:
Yes. It is attached.