BhallaLab/MouseBehaviour #33: data representation- Summary plots

soumyaaymous, 2019-03-20:
would be useful to make the summary plots look like this.
[image]

soumyaaymous, 2019-03-20:
1)  hotter colour used to show eye closure  
2) CS and US showed as little bumps below the main plot. 
3) All plots aligned in the time axis
4)we can keep the current convention of 0 being the CS onset.
5) Raw trace for all trials (including probes ) shown in the same plot..

dilawar, 2019-03-21:
- [ ] Colormap is `jet` (I guess). Not colorblind friendly. Currently we use `viridis` but its upto you.
- [ ] Alignment.
- [ ] Probe are the part of single trial.

dilawar, 2019-03-21:
[image]

Current status.

dilawar, 2019-03-21:
Not sure why timestamp don't match in our and ananth's plot.

soumyaaymous, 2019-03-21:
He doesn't plot CS onset at 0ms. That's alright.

dilawar, 2019-03-22:
[image]

dilawar, 2019-07-07:
Needs to refresh this issue @soumyaaymous . I now forgot what is lacking.

dilawar, 2019-07-18:
On master branch, there is `gui.py`. Launch it and try to do analysis. See the documentation here: https://bhallalab.github.io/MouseBehaviour/