BhallaLab/MouseBehaviour #37: Allocate free pin in Arduino to second US

hrishi16, 2019-04-03:
There is one set of free pins that needs to be allocated to foot shock. Along with that, the protocol code to get only CS and  US2

dilawar, 2019-04-04:
Commit 51fb6eb485e3f05df5355f80e39e358a090332e9 assign pin 10 to shock. Also this pin 
value is written to data as following.

```
  sprintf(msg, "%lu,%s,%d,%d,%d,%d,%d,%d,%d,%s,%ld"
            , timestamp, PROTO_CODE, trial_count_
            , puff, tone, led, shock
            , camera, microscope, trial_state_, encoder_value_
);
```

It is not clear in which protocol this pin suppose to be used. The data will be
printed in all protocol.

See the protocol file
https://github.com/BhallaLab/MouseBehaviour/blob/issue37/Protocols/BehaviourProtocols.csv
as add your protocol to it. Fork/Edit and send a PR. 

Once everything is clear, I'll create a PR/

hrishi16, 2019-04-04:
For now the codes 4 and Hr1 which has puff 2 needs to incorporate the shock, i.e. basically replace puff2 with shock.
I'l need 2 more additional codes like So1 and SoAn1, where the puff1 s replaced by the shock and everythig else is the same. can name it as Hr2 and Hr3

dilawar, 2019-04-04:
That's all fine but its too much of work to maintain multiple version of codes.

You need to add couple of lines in the protocol file with your
requirements. Copy paste the line at the end, change the appropriate
variable and give it appropriate code HRI1 or something similar.

dilawar, 2019-04-23:
I think this is being followed in #39

closed 2019-04-23 by dilawar