BhallaLab/MouseBehaviour #41: Sessions saved over older files

ananthamurthy, 2019-06-01:
I'd tried two sessions with "All1" and found session 2 to have rewritten the trials of "X_1_All1". We need to be sure the session is saved properly

dilawar, 2019-06-04:
If a overwrite is detected (file already exists), the system will STOP and will not overwrite. This is a safety feature implemented in 4ca4fa7

dilawar, 2019-06-04:
Also watch out for camke warnings due to typo in command line parameters e.g. passing `SESSION_TYPE` instead of `SESSION_NUM` will raise the warning but continue with previous value of `SESSION_NUM`.

```
CMake Warning:
  Manually-specified variables were not used by the project:

    SESSION_TYPE
```

Ideally typo in these variables they should be treated as error but currently there is no way.

commit 916fea01 (2019-06-11): How stupid of me. Raise error when file exists.
Start trial from 1.
Fixes #41
closed 2019-06-11 by dilawar