BhallaLab/MouseBehaviour #42: The no-stim protocols have at least air-puff on

ananthamurthy, 2019-06-01:
There were some protocols with no-stim but the air-puff is still turning on at the right time

dilawar, 2019-06-04:
This may happen in PROB trials. If these were not PROBE trials then its a bug. If we don't need PUFF in no-stim in PROBE as well, then I need to change the logic.

ananthamurthy, 2019-06-04:
Hey!

Yes, no need for probe trials in the no-stim or limited stim protocols..

- Ananth

dilawar, 2019-06-04:
Sorry let me be clearer:

- When trial is `PROBE`, then nothing happens in US.
- Else US is played.

I understand when CS is NONE and there should _NOT_ be any US (puff/tone) played? Is that right?

dilawar, 2019-06-04:
@ananthamurthy ping.

ananthamurthy, 2019-06-04:
Hey,
Yes, no cs means no puff either.

Additionally, we could have a specific protocol with only puff.

dilawar, 2019-06-04:
I think the new commit on master should fix it. I am leaving this open, needs testing. I'll it close once you confirm that things are fine.

dilawar, 2019-06-10:
They are in config.h file.
```
/*-----------------------------------------------------------------------------
 *  PINS
 *-----------------------------------------------------------------------------*/
#define         ROTARY_ENC_A                2   // have callback
#define         ROTARY_ENC_B                3   // have callback.
#define         SHOCK_PWM_PIN               5   // PWM
#define         SHOCK_RELAY_PIN             6   
#define         SHOCK_STIM_ISOLATER_PIN     7
#define         TONE_PIN                    8
#define         LED_PIN                     9
#define         CAMERA_TTL_PIN              10
#define         PUFF_PIN                    11
#define         IMAGING_TRIGGER_PIN         13
#define         SHOCK_PAD_READOUT_PIN       A3
#define         SENSOR_PIN                  A5

```

dilawar, 2019-07-07:
Is there any update on this @ananthamurthy ? Just pinging to refresh the old tickets.