BhallaLab/MouseBehaviour #52: Review of PCM

dilawar, 2019-07-07:
@hrishi16 @ananthamurthy @soumyaaymous @upibhalla 

Have a look at this PCB design: https://labnotes.ncbs.res.in/bhalla/mousebehaviour-cad-files-release-20190707pre2 

Schematic of this release is here: https://github.com/BhallaLab/MouseBehaviour/releases/download/v20190707.pre2/AnimalBehaviour.pdf 

Eyeball the schematic. Let me know if you notice something odd/missing. Tomorrow evening, @mrdorababu is going to drill couple of PCBs.

__IMP__ Required components are here: https://github.com/BhallaLab/MouseBehaviour/releases/download/v20190707.pre2/BOM_MouseBehaviour.xls . This sheet contains component required to make 1 PCB. 

Please place the order as soon as possible for as many PCB as required. My BOM on [mouser.in] is here: 
https://www.mouser.in/ProjectManager/ProjectDetail.aspx?AccessID=5a1313ee2d  . If you have mouser accounts, I can share the project with you guys.

dilawar, 2019-07-08:
Notes from @mrdorababu

1. The diode which is connected in the solenoid driver need to be in reverse bias. So we need to connect -ve to the 12V.
2. The AGND and DGND are shorted in the PCB. We need to isolate DGND and AGND.

dilawar, 2019-07-08:
@soumyaaymous @ananthamurthy @hrishi16 reminder. I'd be nice to have one review before it goes to Fab today.

hrishi16, 2019-07-08:
To whatever extent I know, this should be fine.

ananthamurthy, 2019-07-08:
Seems ok.. let's proceed

dilawar, 2019-07-08:
This will go for fab.

dilawar, 2019-07-08:
[image]

Got one made today. I will soldier/mount the components tomorrow morning in workshop. The components needs to be ordered. I've not heard from anyone about order yet. Assuming someone is on it.

dilawar, 2019-07-09:
Release v2019.07.09.rc1 has thicker tracks.

closed 2019-07-09 by dilawar