BhallaLab/MouseBehaviour #57: Modifying protocols

hrishi16, 2019-08-05:
Just adding a line or changing it in the local '.csv' file in 'Protocols' foldr does not start the new protocol. This is the error that comes:

[image]

dilawar, 2019-08-06:
Can you copy paste everything in terminal and upload as a text file. The error occured somewhere way up during cmakr phase.

dilawar, 2019-08-06:
https://unix.stackexchange.com/q/200637/5362

hrishi16, 2019-08-06:
This doesn't work with the docker terminal. The screenshot above is from the docker gui, couldn't find a way to write it. I also tried executing command in docker using this https://www.edureka.co/community/10588/how-can-i-run-a-docker-exec-command-inside-a-docker-container   .... this didn't exactly run in the docker but the main terminal.

dilawar, 2019-08-06:
Docker fetches code from github; so unless things are committed to master branch, your local changes won't have any effect inside docker. I'll add changes to make sure that local changes are visible inside docker.

hrishi16, 2019-08-07:
[100%] Built target cam_server
+ cd ..
+ python3 gui.py --build-dir /root/MouseBehaviour/_docker_build
Traceback (most recent call last):
  File "gui.py", line 25, in <module>
    from screeninfo import get_monitors
  File "/usr/local/lib/python3.6/dist-packages/screeninfo/__init__.py", line 1, in <module>
    from .common import Enumerator, Monitor
  File "/usr/local/lib/python3.6/dist-packages/screeninfo/common.py", line 3, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'
Makefile:14: recipe for target 'run' failed
make: *** [run] Error 1

New error after yesterdays  update

dilawar, 2019-08-08:
screeninfo latest version was released on July 28 which is not compatible with 3.6. it requires dataclass. I'll pin the version to previous release to fix it.

closed 2019-08-08 by dilawar