BhallaLab/MouseBehaviour #61: Add protocol for delay conditions

hrishi16, 2019-08-27:
Changing the present csv doest take care of this.
[image]

A protocol format neds to be added such that the trace interval is not there, and there is an overlap of the 2 stimuli (CS and US. They co-terminate.
Variations can be made in duration of the stimuli.

dilawar, 2019-08-29:
This is possible without using timers as well. I'll send a PR over the weekend.

dilawar, 2019-09-26:
Will there be PROBE trial in this? If there is any PROBE, What is the behaviour of PROBE?

hrishi16, 2019-09-26:
There will be probe. The probe trial will only have the CS (tone or light)
and no US

dilawar, 2019-09-26:
In current revision, following is allowed.

Completely overlapping CS/US: Allowed.

```:
   +----------------------------------------------+
   |                +-----------------------------+                                       
   |                |                             |
--+ ----------------+                             +-------------------
```
US is completely inside CS.: Allowed

```
   +-------------------------------------------------------+
   |                +-----------------------------+        |
   |                |                             |        |
--+ ----------------+                             +--------+----------
```

US is not completely inside CS: __NOT ALLOWED__.

```
   +-------------------------------------+
   |                +--------------------|---------+           
   |                |                    |         |            
--+ ----------------+                    +---------+------------------
```
If the last one is required, this will require a major rewrite/refactoring. If not, then we are almost done.

dilawar, 2019-09-26:
> There will be probe. The probe trial will only have the CS (tone or light) and no US
> […](#)

Ok. PROBE is allowed.

dilawar, 2019-09-27:
@hrishi16 Have a look when  you get time.

dilawar, 2019-10-09:
```
$ cd MouseBehaviour
$ git pull
$ git branch #to see which branch you are on; mentioned by * in front of it.
$ git checkout issue_61   #branch for issue 61 
```
Rest is the same.