BhallaLab/MouseBehaviour #68: no module named cv2

hrishi16, 2020-09-12:
I tried many versions of installing opencv using pip and python, according to their docs. None of them working. 
This  is error in when running analysis file:

"
hrishi@hrishi-Latitude-E5270:~/Dilawar_MB/MouseBehaviour/analysis$ python analyze_trial.py --datadir=/home/../../../ --outdir=/../../analysis

Traceback (most recent call last):
  File "analyze_trial.py", line 13, in <module>
    import analyze_trial_video
  File "/home/hrishi/Dilawar_MB/MouseBehaviour/analysis/analyze_trial_video.py", line 17, in <module>
    import cv2
ImportError: No module named cv2
"

Already tried updating scikit, numpy, and setuptools.

 
Some error while trying to install opencv :
"
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-moeln923/opencv-python/
"

It started when i tried to install ./requirements.txt in the analysis folder:
"
Collecting opencv-python (from -r ./requirements.txt (line 5))
  Downloading https://files.pythonhosted.org/packages/77/f5/49f034f8d109efcf9b7e98fbc051878b83b2f02a1c73f92bbd37f317288e/opencv-python-4.4.0.42.tar.gz (88.9MB)
    100% |████████████████████████████████| 88.9MB 12kB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-lh0ewe0h/opencv-python/setup.py", line 9, in <module>
        import skbuild
    ModuleNotFoundError: No module named 'skbuild'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-lh0ewe0h/opencv-python/
"

hrishi16, 2020-09-12:
@dilawar the readme file is incomplete too.

dilawar, 2020-09-12:
Python3, python is probably python2 on 18.04.

hrishi16, 2020-09-12:
I tried all the versions python and python3 commands, pip -m commands, still the issue is there

hrishi16, 2020-09-15:
I figure the issue. Silly mistake.

closed 2020-09-15 by hrishi16